orocos/rtt_ros_integration #47: typegen executable missing

gborghesan, 2015-10-21:
Hi, i have to build old component with typekit (14.04, OROCOS Toolchain version '2.8.1' from deb)
it seems that the binaries of orogen and typegen are missing.

with a make in the kuka_lwr_fri package i get
```
...
CMake Error at /opt/ros/indigo/lib/cmake/orocos-rtt/UseOROCOS-RTT.cmake:547 (message):
  'typegen' not found in path.  Can't build typekit.  Did you 'source env.sh' ?
...
```
if i try to 
```
$ locate typegen
/opt/ros/indigo/lib/rtt_ros/typegen
```

that contains only 
```
$ cat /opt/ros/indigo/lib/rtt_ros/typegen
#!/usr/bin/env bash

# Wrapper script 

typegen "$@"
```

smits, 2015-10-22:
Can you try `which typegen`

R.

gborghesan, 2015-10-22:
hi ruben,  there is no typegen executable in the path (as i said, it seems that there is no executable at all)
```
$ which typegen
$
```
my colleague that has 12.04 and groovy, has typegen under the orogen folder (i tried yesterday)