orocos/rtt_ros_integration #59: Cleaning up registered services

guihomework, 2016-05-25:
I am interested in dynamically loading and unloading components.
This works nice for standard components with ```d:loadComponent()``` and ```d:unloadComponent()```, but when a ```ROSServiceService``` was loaded into a component, the service proxy seems to not "unadvertise" correctly on component unload and throws errors if the same service "topic" is used.

For instance, I added some controller manager orocos_services to a component by following this example https://github.com/jbohren/conman/blob/master/conman_ros/src/ros_interface_service.cpp  
This orocos_service uses the ```ROSServiceService``` to connect, meaning that it advertises a ros service through a ```NodeHandle```.

However on unload, there is no disconnect in ```ROSServiceService``` and the ```NodeHandle``` apparently does not shutdown correctly so that I get errors if I redeploy the same component+orocos_service.

```
[ERROR] [1464168075.512551730, 1005.620000000]: Tried to advertise a service that is already advertised in this node [/ra/controller_manager/list_controller_types]
```
even if the component does not appear anymore (neither in ```rttlib.stat()```  nor in ```d:getPeers()```)

You might argue that a controller manager should always be there even in the case of dynamic deployment (conman does that by managing the list of peers and such), but the problem goes beyond the functionality of a controller manager, any orocos_service could create a ```ROSServiceService``` that at some point might needs to be cleaned-up for redeployment later on when using dynamic deployment. 

What would be a correct way to unadvertise the service ? at which level should it happen ? I could then try to modify the ```rtt_roscomm``` to handle that and create a PR.

meyerj, 2016-05-25:
This issue could probably be resolved by replacing the bare pointers in the value type of the maps holding the service proxies with shared pointers in [rtt_rosservice_service.cpp:161](https://github.com/orocos/rtt_ros_integration/blob/indigo-devel/rtt_roscomm/src/rtt_rosservice_service.cpp#L161). Another alternative is that the destructor of the `ROSServiceService` class cleans up and deletes the proxies.

guihomework, 2016-05-30:
Sounds like something I can try easily. thanks for the tips

guihomework, 2016-06-01:
Issue can be closed not that #60 is merged

closed 2016-06-01 by guihomework