orocos/rtt_ros_integration #68: rtt_tf: migrate to tf2

meyerj, 2016-07-19:
The current [rtt_tf](https://github.com/orocos/rtt_ros_integration/tree/indigo-devel/rtt_tf) package is not fully compatible to [tf2](http://wiki.ros.org/tf2) used in ROS hydro and upwards. We need to [migrate](http://wiki.ros.org/tf2/Migration) or clearly mark it as deprecated.

meyerj, 2019-04-18:
### TODO

- use [tf2::BufferCore](https://github.com/ros/geometry2/blob/melodic-devel/tf2/include/tf2/buffer_core.h) instead of `tf::Transformer`
- add a `rtt_tf2_msgs` typekit to [typekits/](https://github.com/orocos/rtt_ros_integration/tree/toolchain-2.9/typekits)
- use tf2_msgs/TFMessage for the `tf` input port, and add a second input port for `tf_static`

francisco-miguel-almeida, 2019-04-29:
Ported to tf2 in branch: https://github.com/orocos/rtt_ros_integration/pull/121

closed 2019-05-15 by meyerj