orocos/rtt_ros_integration #77: create_rtt_msgs missing dep

ahoarau, 2016-09-09:
If I create a pkg `foo_msgs`, and the `rtt_foo_msgs` ( generated by `rosrun rtt_roscomm create_rtt_msgs foo_msgs`), then if clean and rebuild (`catkin clean && catkin build`), I'm having errors like : 

```cpp
... /.private/rtt_foo_msgs/include/orocos/foo_msgs/boost/MyAction.h:9:10: fatal error: 'foo_msgs/MyAction.h' file not found
#include <foo_msgs/MyActionFeedback.h>
```

Shouldn't we add catkin message dep + include dirs + depend on generated target ?

```cmake
+ find_package(catkin REQUIRED COMPONENTS rtt_roscomm foo_msgs)

+ include_directories(${catkin_INCLUDE_DIRS})

orocos_generate_package(
  DEPENDS foo_msgs
+  DEPENDS_TARGETS  ...  foo_msgs_generate_messages_cpp
```

meyerj, 2016-09-09:
`orocos_generate_package(DEPENDS ...)` and  `orocos_generate_package(DEPENDS_TARGETS ...)` expect either pkg-config or Orocos package names, not cmake targets. The term "target" refers to Orocos targets (like gnulinux, xenomai, macos, etc.). And the dependencies declared in `orocos_generate_package()` only influence other packages depending on the `rtt_foo_msgs`, not the package itself. So this is neither the problem nor the solution.

Normally the variable `${_package}_EXPORTED_TARGETS` in [templates/typekit/CMakeLists.txt:138](https://github.com/orocos/rtt_ros_integration/blob/indigo-devel/rtt_roscomm/src/templates/typekit/CMakeLists.txt#L138) should contain the `*_generate_messages_cpp` target which is added as a dependency for Boost header generation and all the libraries. Not sure while this mechanism does not work for you.

Are you sure that the problem is indeed related to wrong ordering of make targets during the build process or is package `foo_msgs` simply not exporting the generated headers for some reason?

ahoarau, 2016-09-09:
> orocos_generate_package(DEPENDS ...) and orocos_generate_package(DEPENDS_TARGETS ...) expect either pkg-config or Orocos package names, not cmake targets. The term "target" refers to Orocos targets (like gnulinux, xenomai, macos, etc.). 

Then who gets the foo_msgs include dirs if not `find_package(catkin REQUIRED COMPONENTS rtt_roscomm foo_msgs)` and  `include_directories(${catkin_INCLUDE_DIRS})` ?

> And the dependencies declared in orocos_generate_package() only influence other packages depending on the rtt_foo_msgs, not the package itself. So this is neither the problem nor the solution.

Agreed for this one.

--------
For the example, here's my pure ROS msg pkg : https://github.com/kuka-isir/krl_msgs/blob/master/CMakeLists.txt

> Are you sure that the problem is indeed related to wrong ordering of make targets during the build process or is package foo_msgs simply not exporting the generated headers for some reason?

The ordering seems fine, the `foo_msgs` does not seem to export the headers correctly, or `rtt_foo_msgs`does not seems to get them :-)

ahoarau, 2016-09-10:
Please note that if the package has been built before, the rtt wrapper builds fine also.

ahoarau, 2016-09-27:
I tried with a test package from scratch and could not reproduce the error. Closing until I figure out why this package is not building correctly. Thanks !

closed 2016-09-27 by ahoarau