orocos/rtt_ros_integration #78: kinetic release

Timple, 2016-11-12:
The source compiles fine under ubuntu 16.04 with ROS kinetic.

Only dependencies where liblua5.1.0-dev and libncurses5-dev.

Would it be possible to release the binaries for ROS kinetic?

jlack1987, 2017-04-27:
Agreed! Thought it was kinda strange there is a release for indigo and jade but no kinetic. Been waiting on this for a few months now

meyerj, 2017-05-05:
I did a first pre-release of RTT, OCL, rtt_ros_integration and rtt_geometry into ROS kinetic this week. But there are still some pending and potentially API breaking changes to be merged into RTT, especially https://github.com/orocos-toolchain/rtt/pull/206.

After the final version of 2.9 has been released and announced we are planning to also release the exact same versions into ROS lunar. I have not tried it yet.

However, as the kinetic release is basically done, I am going to close this issue.

closed 2017-05-05 by meyerj

ahoarau, 2017-05-05:
Excellent news !