orocos/rtt_ros_integration #87: Bug in create_rtt_msgs (rtt_roscomm) when target to services

haianos, 2017-04-22:
For some reason, there are visibility issues (maybe something wrong in the CMake macros..?) when generating a typekit for ROS services with `create_rtt_msgs` tool.
Everything works fine if the target package contains .msg, but the issue affects only  `.srv`.
The generated package `rtt_<target_pkg>` properly create the line `ros_generate_rtt_service_proxies`, but in the user code of the component the header file is not found (eg, `#include <target_pkg/MyService.h>`).
The issue is not on the component/package using the `rtt_<target_pkg>`, since the same user code uses `rtt_std_srvs`, which works just fine.

For now, the solution (apparently) is to include the dependency of "rosgraph_msgs" in the CMakeLists.txt as follows (but I can't explain why this solve the issue):
```
orocos_generate_package(
  DEPENDS rosgraph_msgs
  DEPENDS_TARGETS rtt_roscomm rtt_std_msgs
)
```
This dependency is mentioned in the `rtt_std_srvs` (which works fine), but `create_rtt_msgs` does not include it.

Possible actions:
1) if the solution is correct, update the `create_rtt_msgs` by adding that dependency;
2) otherwise, the cause could be somewhere else...

NOTE: Tested on Ubuntu 16.04 LTS, v2.8.3 from sources (maybe the issue is already solved in latest version).

ahoarau, 2017-04-22:
I think I had the same issue some time ago : https://github.com/orocos/rtt_ros_integration/issues/77
My colleague @jimmydasilva encountered this last month.
Same for https://github.com/jbohren/conman/commit/b26d008551bd0c1e168015d2274ec8f09d3a3593

haianos, 2017-04-22:
Yes, they might be related issues. In fact, I am not sure whether my solution _is_ the solution, but it is possible that adding the `rosgraph_msgs` dependency triggers something in the cmake compilation process that clean/fix the real issue...

haianos, 2017-04-22:
I've double checked by removing the dependency from `rosgraph_msgs`, and as suspected 1) now it keeps compiling fine (so there is no real need for that dependency), but 2) that change triggered something in the compilation process (so it could be an useful hint)

ahoarau, 2017-04-22:
The fix that I found was to add the package name in the catkin components (as in jbohren/conman@b26d008), could you try this ?

haianos, 2017-04-22:
I tried, and it does not give any additional value (but right now, the issues is also fixed).
I think that "my fix" or that fix do not really solve the problem (because they are not necessary), but they do trigger something internal that makes everything compiling fine.

meyerj, 2017-05-02:
Could you please check whether #89 fixes the issue?

haianos, 2017-05-02:
Thanks Johannes!
Checking the description, it sounds like a solution! However, I will be able to test it out only in 2 weeks...

meyerj, 2017-05-02:
Okay, I'll do a pre-release in kinetic including the patch. We can merge and close this issue after you confirmed that the patch works for you.

commit 02f67086 (2017-05-17): rtt_roscomm: fixed missing package headers include directories for service proxies (fix #87)

If the service package is not in the same workspace as the corresponding RTT service proxy package or
depends on other packages in underlaying workspaces, include directory ${CAKTIN_DEVEL_PREFIX}/include
is not sufficient. Instead, all directories found by find_package() must be added as include directories,
like for typekits.
closed 2017-05-17 by meyerj