orocos/rtt_ros_integration #91: Do rosservice calls respect orocos operation thread spec?

jlack1987, 2017-07-30:
Say I have an orocos component that has an operation such as,

```
this->addOperation("do_work", &MyWorker::doWork, this, RTT::ClientThread).doc("Does work");
```
Where when called, this **blocks the callers thread** and **does not block the task context thread**. I'm unable to find anywhere in the documentation for `rtt_ros_integration` whether or not `ROSServiceService` calls respect the orocos operation spec on whether to block the client thread of the task context thread.

gavanderhoorn, 2017-07-31:
Related: [Do rosservice calls from rtt_ros respect orocos operation thread spec?](http://answers.ros.org/question/267680) on ROS Answers.

meyerj, 2017-07-31:
The ROS service operation does not behave differently than any other RTT operation here: As you have added it as `RTT::ClientThread`, the caller (the ROS spinner in this case) will execute the operation in its own thread without synchronization with the component's activity. In most cases you therefore want to add the operation with `RTT::OwnThread`, in which case the component's activity/thread will execute it. So the short answer is: Yes, the spec is respected.

It should also be noted that the ROS spinner will always be blocked and eventually waits for the successful completion of the operation, even with `RTT::OwnThread`. If you are using [rtt_rosnode](https://github.com/orocos/rtt_ros_integration/tree/toolchain-2.9/rtt_rosnode) you can increase the number of spinner threads if you do not want to block incoming messages or service calls (see https://github.com/orocos/rtt_ros_integration/commit/e6a57dc3c8caf9e517800e85fdc93d541c26f050) until the operation is done.

We could indeed add a note to the rtt_roscomm documentation in [rtt_roscomm/README.md](https://github.com/orocos/rtt_ros_integration/blob/toolchain-2.9/rtt_roscomm/README.md) and also document the `~spinner_threads` parameter or even increase the default to 4 or more.

References:
- [addOperation() API](https://orocos-toolchain.github.io/rtt/master/api/html/classRTT_1_1Service.html#a9f467780775bea590fcb6e6eef72b580)
- [Component Builder's Manual](https://orocos-toolchain.github.io/rtt/master/xml/orocos-components-manual.html#execution-types-table)

jlack1987, 2017-07-31:
Thanks, the behavior i'm going for is I would like a `TaskContext` to to have some `Operation`s that are made available through ROS service calls, but I don't want the service call to block the thread calling the `updateHook()` function. Long story short I think you said the orocos component can create services and those services **will not** block the components calling of `updateHook()`.

ahoarau, 2017-07-31:
You can create components, and connect the operations to ROS services without having the `updateHook()` blocked, using `RTT::ClientThread`. In the end it's the `rosservice` that calls the operations and transmits them to ROS, so it's `rosservice` the `Client` in that case.

jlack1987, 2017-08-02:
Just to try to add some to this. I did end up getting this working, but had to look at what was done in the [unit test](https://github.com/orocos/rtt_ros_integration/blob/toolchain-2.9/rtt_roscomm/src/rtt_rosservice_service.cpp) to figure out how to get it all working properly. Some more thorough examples for how to get started using the awesome features here would be great and probably extremely helpful for some who won't crawl through the code to figure it out.