orocos/rtt_ros_integration #95: 2.9.1 failing to build on Trusty armhf

tfoote, 2017-11-16:
@meyerj https://github.com/ros/rosdistro/pull/16385 appears to have caused a regression on Trusty armhf: http://build.ros.org/view/Ibin_arm_uThf/job/Ibin_arm_uThf__rtt_roscomm__ubuntu_trusty_armhf__binary/

```
00:25:50.966 -- Found PkgConfig: /usr/bin/pkg-config (found version "0.26") 
00:25:50.990 -- [UseOrocos] Building package rtt_roscomm with catkin develspace support.
00:25:50.991 -- [UseOrocos] Using Orocos RTT in rtt_roscomm
00:25:51.367 -- checking for one of the modules 'rtt_ros-gnulinux'
00:25:52.230 -- [UseOrocos] Found orocos package 'rtt_ros'.
00:25:52.260 -- checking for one of the modules 'rtt_rospack-gnulinux'
01:20:54.950 FATAL: command execution failed
01:20:54.950 org.jenkinsci.remoting.nio.FifoBuffer$CloseCause: Buffer close has been requested
01:20:54.951 	at org.jenkinsci.remoting.nio.FifoBuffer.close(FifoBuffer.java:426)
01:20:54.951 	at org.jenkinsci.remoting.nio.NioChannelHub$MonoNioTransport.closeR(NioChannelHub.java:332)
01:20:54.951 	at org.jenkinsci.remoting.nio.NioChannelHub.run(NioChannelHub.java:565)
01:20:54.951 Caused: java.io.IOException: Unexpected EOF while receiving the data from the channel. FIFO buffer has been already closed
```
http://build.ros.org/view/Ibin_arm_uThf/job/Ibin_arm_uThf__rtt_roscomm__ubuntu_trusty_armhf__binary/13/consoleFull
http://build.ros.org/view/Ibin_arm_uThf/job/Ibin_arm_uThf__rtt_roscomm__ubuntu_trusty_armhf__binary/18/console

The buildfarm has retried several times and basically times out during the cmake configure step here.

tfoote, 2017-11-16:
@nuclearsandwich FYI

meyerj, 2017-11-17:
Hmm, I don't see how this is related to one of the minimal changes and why the problem would only affect armhf builds. I also checked the contents of http://repositories.ros.org/ubuntu/building/pool/main/r/ros-indigo-rtt-rospack/ros-indigo-rtt-rospack_2.8.6-0trusty-20171116-074906-0800_armhf.deb and they look okay.

Orocos cmake macros call `pkg-config` under the hood, which opens a pipe between `pkg-config` and cmake. Are the ARM builds running with qemu? We also had problems in the past with cmake and qemu, similar to https://github.com/ros-infrastructure/buildfarm/issues/87. Problem seems to be related, because the FATAL error happens almost an hour after the last line, so probably the job is killed due to a timeout.

meyerj, 2017-11-20:
This issue apparently has been solved since build [#22](http://build.ros.org/view/Ibin_arm_uThf/job/Ibin_arm_uThf__rtt_roscomm__ubuntu_trusty_armhf__binary/22/) succeeded?