orocos/rtt_ros_integration #105: issue with orocos_rtt / rtt_roscomm after migrating to ros kinetic

onilsson94, 2018-07-02:
Hello, I have an issue with a project using orocos modules

I am currently trying to compile a project from Ubuntu 14.04 in 16.04 and have had to fix quite a few problems. This probem seems to be related to orocos for ros, but I’m unable to find the origin of the problem, so I have mostly made random searches with key words I thought to be relevant.

I am posting here because someone sugested that it may be an issue with orocos_rtt / rtt_roscomm on a post I made on the ros forum [here](https://answers.ros.org/question/295911/migrating-from-ros-indigo-to-ros-kinetic-cmake-cant-finish-the-run/?comment=295923#post-id-295923)

For this project I am using these orocos packages for ros: rtt_ros_integration, rtt_common_msgs, rtt_ros_comm and rtt-roscomm and it is constructed with catkin

Does anyone recognize this problem?

Here is the error generated by cmake:

```
-- Checking for one of the modules 'rtt_rosgraph_msgs-gnulinux'
-- [UseOrocos] Found orocos package 'rtt_rosgraph_msgs'.
-- Checking for one of the modules 'rtt_geometry_msgs-gnulinux'
-- [UseOrocos] Found orocos package 'rtt_geometry_msgs'.
-- - Detected OROCOS_TARGET environment variable. Using: gnulinux
-- Found orocos-rtt 2.9.0 for the gnulinux target. Available transports: corba mqueue
-- - Found requested orocos-rtt components: rtt-marshalling rtt-scripting rtt-scripting rtt-marshalling rtt-transport-mqueue rtt-transport-corba
-- Using these message generators: gencpp;geneus;genlisp;gennodejs;genpy
-- ros_generate_rtt_typekit: Could not find any .msg files in the lc_control package.
-- Using these message generators: gencpp;geneus;genlisp;gennodejs;genpy
-- [UseOrocos] Building plugin library rtt_lc_control_rosservice_proxies
CMake Error at /opt/ros/kinetic/lib/cmake/orocos-rtt/UseOROCOS-RTT.cmake:761 (target_link_libraries):
  Target "lc_control_generate_messages_eus" of type UTILITY may not be linked
  into another target.  One may link only to STATIC or SHARED libraries, or
  to executables with the ENABLE_EXPORTS property set.
Call Stack (most recent call first):
  /opt/ros/kinetic/lib/cmake/orocos-rtt/UseOROCOS-RTT.cmake:786 (orocos_plugin)
  /opt/ros/kinetic/share/rtt_roscomm/src/templates/service/CMakeLists.txt:93 (orocos_service)

CMake Error at /opt/ros/kinetic/lib/cmake/orocos-rtt/UseOROCOS-RTT.cmake:761 (target_link_libraries):
  Target "lc_control_generate_messages_nodejs" of type UTILITY may not be
  linked into another target.  One may link only to STATIC or SHARED
  libraries, or to executables with the ENABLE_EXPORTS property set.
Call Stack (most recent call first):
  /opt/ros/kinetic/lib/cmake/orocos-rtt/UseOROCOS-RTT.cmake:786 (orocos_plugin)
  /opt/ros/kinetic/share/rtt_roscomm/src/templates/service/CMakeLists.txt:93 (orocos_service)

CMake Error at /opt/ros/kinetic/share/rtt_roscomm/src/templates/service/CMakeLists.txt:94 (target_link_libraries):
  Target "lc_control_generate_messages_eus" of type UTILITY may not be linked
  into another target.  One may link only to STATIC or SHARED libraries, or
  to executables with the ENABLE_EXPORTS property set.

CMake Error at /opt/ros/kinetic/share/rtt_roscomm/src/templates/service/CMakeLists.txt:94 (target_link_libraries):
  Target "lc_control_generate_messages_nodejs" of type UTILITY may not be
  linked into another target.  One may link only to STATIC or SHARED
  libraries, or to executables with the ENABLE_EXPORTS property set.

-- [UseOrocos] Generating package version 0.0.0 from rtt_lc_control_VERSION (package.xml).
-- [UseOrocos] Generating pkg-config file for package in catkin devel space.
-- [UseOrocos] Exporting targets rtt_lc_control_rosservice_proxies;roscpp_generate_messages_cpp;roscpp_generate_messages_eus;roscpp_generate_messages_lisp;roscpp_generate_messages_nodejs;roscpp_generate_messages_py;rosgraph_msgs_generate_messages_cpp;rosgraph_msgs_generate_messages_eus;rosgraph_msgs_generate_messages_lisp;rosgraph_msgs_generate_messages_nodejs;rosgraph_msgs_generate_messages_py;std_msgs_generate_messages_cpp;std_msgs_generate_messages_eus;std_msgs_generate_messages_lisp;std_msgs_generate_messages_nodejs;std_msgs_generate_messages_py.
-- [UseOrocos] Exporting libraries rtt_lc_control_rosservice_proxies;lc_control_generate_messages_eus;lc_control_generate_messages_nodejs;Orchestrator;BaseRamper;BaseSampler;MobilePID;MecanumIK;MecanumIK_rosHwi;MecanumIK_realHwi;MecanumFK;MecanumFK_realHwi;MecanumFK_rosHwi;MobileStatus_rosHwi;TrayPID;ScrewIK;ScrewIK_rosHwi;ScrewIK_realHwi;ScrewFK;ScrewFK_rosHwi;ScrewFK_realHwi;TrayStatus_rosHwi;ArmSampler;ArmIK;ArmIK_rosHwi;ArmIK_realHwi;ArmFK_realHwi;/opt/ros/kinetic/lib/librtt_rostopic-gnulinux.so;/opt/ros/kinetic/lib/orocos/gnulinux/rtt_roscomm/plugins/librtt_rostopic_service-gnulinux.so;/opt/ros/kinetic/lib/orocos/gnulinux/rtt_roscomm/plugins/librtt_rosservice_registry-gnulinux.so;/opt/ros/kinetic/lib/orocos/gnulinux/rtt_roscomm/plugins/librtt_rosservice-gnulinux.so;/opt/ros/kinetic/lib/liborocos-rtt-gnulinux.so;/usr/lib/x86_64-linux-gnu/librt.so;/opt/ros/kinetic/lib/libroscpp.so;/usr/lib/x86_64-linux-gnu/libboost_filesystem.so;/usr/lib/x86_64-linux-gnu/libboost_signals.so;/opt/ros/kinetic/lib/librosconsole.so;/opt/ros/kinetic/lib/librosconsole_log4cxx.so;/opt/ros/kinetic/lib/librosconsole_backend_interface.so;/usr/lib/x86_64-linux-gnu/liblog4cxx.so;/usr/lib/x86_64-linux-gnu/libboost_regex.so;/opt/ros/kinetic/lib/libroscpp_serialization.so;/opt/ros/kinetic/lib/libxmlrpcpp.so;/opt/ros/kinetic/lib/librostime.so;/opt/ros/kinetic/lib/libcpp_common.so;/usr/lib/x86_64-linux-gnu/libboost_system.so;/usr/lib/x86_64-linux-gnu/libboost_thread.so;/usr/lib/x86_64-linux-gnu/libboost_chrono.so;/usr/lib/x86_64-linux-gnu/libboost_date_time.so;/usr/lib/x86_64-linux-gnu/libboost_atomic.so;/usr/lib/x86_64-linux-gnu/libpthread.so;/usr/lib/x86_64-linux-gnu/libconsole_bridge.so;/opt/ros/kinetic/lib/orocos/gnulinux/rtt_std_msgs/plugins/librtt-ros-primitives-transport-gnulinux.so;/opt/ros/kinetic/lib/orocos/gnulinux/rtt_std_msgs/types/librtt-std_msgs-typekit-gnulinux.so;/opt/ros/kinetic/lib/orocos/gnulinux/rtt_std_msgs/types/librtt-std_msgs-ros-transport-gnulinux.so;/opt/ros/kinetic/lib/orocos/gnulinux/rtt_rosgraph_msgs/types/librtt-rosgraph_msgs-typekit-gnulinux.so;/opt/ros/kinetic/lib/orocos/gnulinux/rtt_rosgraph_msgs/types/librtt-rosgraph_msgs-ros-transport-gnulinux.so;/opt/ros/kinetic/lib/orocos/gnulinux/rtt_geometry_msgs/types/librtt-geometry_msgs-typekit-gnulinux.so;/opt/ros/kinetic/lib/orocos/gnulinux/rtt_geometry_msgs/types/librtt-geometry_msgs-ros-transport-gnulinux.so.
-- [UseOrocos] Exporting include directories /opt/ros/kinetic/include/orocos;/opt/ros/kinetic/include;/usr/include;/opt/ros/kinetic/share/xmlrpcpp/cmake/../../../include/xmlrpcpp.
-- [UseOrocos] Exporting library directories /home/onilsson/DevRoot/src/catkin_ws/install/lib/orocos/gnulinux/lc_control;/opt/ros/kinetic/lib;/opt/ros/kinetic/lib/orocos/gnulinux/rtt_roscomm/plugins;/opt/ros/kinetic/lib/orocos/gnulinux/rtt_std_msgs/plugins;/opt/ros/kinetic/lib/orocos/gnulinux/rtt_std_msgs/types;/opt/ros/kinetic/lib/orocos/gnulinux/rtt_rosgraph_msgs/types;/opt/ros/kinetic/lib/orocos/gnulinux/rtt_geometry_msgs/types.
-- +++ processing catkin package: 'rtt_actionlib_examples'
-- ==> add_subdirectory(rtt_ros_examples/rtt_actionlib_examples)
-- - Detected OROCOS_TARGET environment variable. Using: gnulinux
-- Orocos-RTT found in /opt/ros/kinetic/lib/cmake/orocos-rtt/orocos-rtt-gnulinux-libraries.cmake
-- Found orocos-rtt 2.9.0 for the gnulinux target. Available transports: corba mqueue
-- - Found requested orocos-rtt components: rtt-marshalling rtt-scripting
-- [UseOrocos] Building package rtt_actionlib_examples with catkin develspace support.
-- [UseOrocos] Using Orocos RTT in rtt_actionlib_examples
-- Checking for one of the modules 'ocl-gnulinux'
-- [UseOrocos] Found orocos package 'ocl'.
-- Checking for one of the modules 'rtt_ros-gnulinux'
-- [UseOrocos] Found orocos package 'rtt_ros'.
-- Checking for one of the modules 'rtt_rosclock-gnulinux'
-- [UseOrocos] Found orocos package 'rtt_rosclock'.
[... 5 lines cut ...]
-- [UseOrocos] Found orocos package 'rtt_actionlib_msgs'.
-- Checking for one of the modules 'roscpp-gnulinux'
-- Checking for one of the modules 'actionlib-gnulinux'
-- Checking for one of the modules 'actionlib_msgs-gnulinux'
-- Checking for one of the modules 'message_generation-gnulinux'
-- Using these message generators: gencpp;geneus;genlisp;gennodejs;genpy
[... cut ...]
-- Checking for one of the modules 'ocl-deployment-gnulinux;ocl-deployment'
-- [UseOrocos] Found orocos package 'ocl-deployment'.
-- Checking for one of the modules 'ocl-taskbrowser-gnulinux;ocl-taskbrowser'
-- [UseOrocos] Found orocos package 'ocl-taskbrowser'.
-- [UseOrocos] Building executable test_server
-- Configuring incomplete, errors occurred!
See also "/home/onilsson/DevRoot/src/catkin_ws/build/CMakeFiles/CMakeOutput.log".
See also "/home/onilsson/DevRoot/src/catkin_ws/build/CMakeFiles/CMakeError.log".
Invoking "cmake" failed

```

meyerj, 2018-07-19:
This bug should be fixed by https://github.com/orocos-toolchain/rtt/pull/244 in combination with https://github.com/orocos/rtt_ros_integration/commit/611d768db465dcd765c22533a8f6888523231129, which are still pending a review and subsequent release.

The previous solution to the problem suggested in https://github.com/orocos-toolchain/rtt/pull/92 only works for a closed list of ROS message generators and associated target names, namely `gencpp`, `genpy` and `genlisp`. Kinetic added `geneus` and `gennodejs` as default message generators, which are not covered by https://github.com/orocos-toolchain/rtt/pull/92.

As a workaround for custom message packages, you can disable the unsupported message generators by setting the `ROS_LANG_DISABLE` environment variable if you do not need them (when compiling the message package `lc_control`, not only for `rtt_lc_control`):

```cpp
export ROS_LANG_DISABLE="geneus:genjava:gennodejs"
```

meyerj, 2019-04-17:
Fixed in https://github.com/orocos-toolchain/rtt/pull/244.

closed 2019-04-17 by meyerj