orocos/rtt_ros_integration #110: rtt_rosnode: loadROSService() overrides one from rtt_ros

mjbogusz, 2018-08-16:
The global function `loadROSService()` in `rtt_rosnode/src/ros_plugin.cpp` added in 6ae4950eae59daf3332cf498be2e38e249365570 apparently overrides one with the same name residing in `rtt_ros/src/rtt_ros_service.cpp` (probably during linking).

This causes the 'import' operation to never be registered thus breaking anything using that operation. Simple renaming the function within the file solved the issue (at least in my usecase) - it's not referenced anywhere outside the file.

---

As an additional note (though I'm not sure about this one), shouldn't it register as providing `"rosnode"` instead of `"ros"`, to be consistent with the name reported in `getRTTPluginName()`?

meyerj, 2018-08-17:
Thanks for reporting! #111 should fix the issue.

> As an additional note (though I'm not sure about this one), shouldn't it register as providing "rosnode" instead of "ros", to be consistent with the name reported in getRTTPluginName()?

It used to be a deliberate choice to collect all operations provided by one of the services in this repo in the `ros` global service (or at least the most common ones), independent of the package name (https://github.com/orocos/rtt_ros_integration/pull/21). RTT does not enforce that the name of the plugin matches the package or service name.

closed 2018-08-17 by meyerj

disRecord, 2018-09-27:
Thanks for the report and the fix! 
Before I proposed this change I had not tried to load `rtt_ronode` and `rtt_ros` together. My bad...

PR #111 (2018-08-17): Declare loadROSService() methods as static to fix name clashes
Fixes #110:
> The global function loadROSService() in rtt_rosnode/src/ros_plugin.cpp added in 6ae4950 apparently overrides one with the same name residing in rtt_ros/src/rtt_ros_service.cpp (probably during linking).
commit 64065117 (2018-08-17): Declare loadROSService() methods as static to fix name clashes (fix #110)
commit b2191e69 (2019-05-15): Declare loadROSService() methods as static to fix name clashes (fix #110)