orocos/rtt_ros_integration #113: Problem to build orocos-rtt 2.8.3 for the gnulinux + ROS Kinetic

MatheusPinto, 2018-09-03:
Hi,

I follow the steps from here (https://github.com/orocos/rtt_ros_integration), to install orocos-rtt in a ROS Kinetic catkin workspace.

Too many things are not clear to me yet:

- in the `cd ~/ws/underlay_isolated`, "ws" is the catkin workspace?
- the packages that I create, must stay in "`catkin_ws/underlay/src`"?
- where I put this commands: `import("rtt_ros") ros.import("my_pkg_name")`?

Assuming the installation is in the catkin directory, I had to install the following packages:

```
apt-get install ruby-dev
gem install rake hoe
sudo apt-get install libreadline-dev 
```

After that, assuming that the package installation should be done in the "`catkin_ws/underlay/src`", I create a package with the same dependencies introduced in rtt_ros_integration doc. However, in a certain point, catkin accuses the following error:

```
-- Checking for one of the modules 'ocl-taskbrowser-gnulinux;ocl-taskbrowser'
CMake Error at /home/matheus/catkin_ws/underlay_isolated/install_isolated/lib/cmake/orocos-rtt/UseOROCOS-RTT-helpers.cmake:208 (message):
  [UseOrocos] Could not find package 'ocl-taskbrowser'.
Call Stack (most recent call first):
  /home/matheus/catkin_ws/underlay_isolated/install_isolated/lib/cmake/orocos-rtt/UseOROCOS-RTT-helpers.cmake:276 (orocos_find_package)
  rtt_ros_integration/rtt_tf/CMakeLists.txt:23 (orocos_use_package)
```

However, it looks like this library is inside the packages:

```
locate ocl-taskbrowser
/home/matheus/catkin_ws/underlay_isolated/build_isolated/ocl/install/taskbrowser/liborocos-ocl-taskbrowser-gnulinux.so
/home/matheus/catkin_ws/underlay_isolated/build_isolated/ocl/install/taskbrowser/liborocos-ocl-taskbrowser-gnulinux.so.2.8
/home/matheus/catkin_ws/underlay_isolated/build_isolated/ocl/install/taskbrowser/liborocos-ocl-taskbrowser-gnulinux.so.2.8.4
/home/matheus/catkin_ws/underlay_isolated/build_isolated/ocl/install/taskbrowser/ocl-taskbrowser-gnulinux.pc
/home/matheus/catkin_ws/underlay_isolated/build_isolated/ocl/install/taskbrowser/CMakeFiles/orocos-ocl-taskbrowser.dir
/home/matheus/catkin_ws/underlay_isolated/build_isolated/ocl/install/taskbrowser/CMakeFiles/orocos-ocl-taskbrowser.dir/CXX.includecache
/home/matheus/catkin_ws/underlay_isolated/build_isolated/ocl/install/taskbrowser/CMakeFiles/orocos-ocl-taskbrowser.dir/DependInfo.cmake
/home/matheus/catkin_ws/underlay_isolated/build_isolated/ocl/install/taskbrowser/CMakeFiles/orocos-ocl-taskbrowser.dir/TaskBrowser.cpp.o
/home/matheus/catkin_ws/underlay_isolated/build_isolated/ocl/install/taskbrowser/CMakeFiles/orocos-ocl-taskbrowser.dir/build.make
/home/matheus/catkin_ws/underlay_isolated/build_isolated/ocl/install/taskbrowser/CMakeFiles/orocos-ocl-taskbrowser.dir/cmake_clean.cmake
/home/matheus/catkin_ws/underlay_isolated/build_isolated/ocl/install/taskbrowser/CMakeFiles/orocos-ocl-taskbrowser.dir/depend.internal
/home/matheus/catkin_ws/underlay_isolated/build_isolated/ocl/install/taskbrowser/CMakeFiles/orocos-ocl-taskbrowser.dir/depend.make
/home/matheus/catkin_ws/underlay_isolated/build_isolated/ocl/install/taskbrowser/CMakeFiles/orocos-ocl-taskbrowser.dir/flags.make
/home/matheus/catkin_ws/underlay_isolated/build_isolated/ocl/install/taskbrowser/CMakeFiles/orocos-ocl-taskbrowser.dir/link.txt
/home/matheus/catkin_ws/underlay_isolated/build_isolated/ocl/install/taskbrowser/CMakeFiles/orocos-ocl-taskbrowser.dir/progress.make
/home/matheus/catkin_ws/underlay_isolated/install_isolated/lib/liborocos-ocl-taskbrowser-gnulinux.so
/home/matheus/catkin_ws/underlay_isolated/install_isolated/lib/liborocos-ocl-taskbrowser-gnulinux.so.2.8
/home/matheus/catkin_ws/underlay_isolated/install_isolated/lib/liborocos-ocl-taskbrowser-gnulinux.so.2.8.4
/home/matheus/catkin_ws/underlay_isolated/install_isolated/lib/pkgconfig/ocl-taskbrowser-gnulinux.pc

```

What ca I do?

ahoarau, 2018-09-03:
https://rtt-lwr.readthedocs.io/en/latest/install/install-16.04-kinetic.html#orocos-2-9-rtt-ros-integration-2-9-from-source

MatheusPinto, 2018-09-03:
@ahoarau unfortunately this did not help me. I did the entire installation process correctly, but at the time of the test an error occurred. When trying to run the gazebo with the ROS plugins, as mentioned in the following text, errors appear:
```

gazebo -s libgazebo_ros_paths_plugin.so -s libgazebo_ros_api_plugin.so --verbose
Gazebo multi-robot simulator, version 9.3.1
Copyright (C) 2012 Open Source Robotics Foundation.
Released under the Apache 2 License.
http://gazebosim.org

[Msg] Waiting for master.
Gazebo multi-robot simulator, version 9.3.1
Copyright (C) 2012 Open Source Robotics Foundation.
Released under the Apache 2 License.
http://gazebosim.org

[Err] [Plugin.hh:180] Failed to load plugin libgazebo_ros_paths_plugin.so: libgazebo_ros_paths_plugin.so: cannot open shared object file: No such file or directory
[Err] [Plugin.hh:180] Failed to load plugin libgazebo_ros_api_plugin.so: libgazebo_ros_api_plugin.so: cannot open shared object file: No such file or directory
[Msg] Waiting for master.
[Msg] Connected to gazebo master @ http://127.0.0.1:11345
[Msg] Publicized address: 192.168.0.106
[Msg] Connected to gazebo master @ http://127.0.0.1:11345
[Msg] Publicized address: 192.168.0.106
[Wrn] [GuiIface.cc:120] void DBusMenuExporterPrivate::addAction(QAction*, int): Already tracking action "&File" under id 56
[Wrn] [GuiIface.cc:120] void DBusMenuExporterPrivate::addAction(QAction*, int): Already tracking action "&Edit" under id 59
[Wrn] [GuiIface.cc:120] void DBusMenuExporterPrivate::addAction(QAction*, int): Already tracking action "&Camera" under id 61
[Wrn] [GuiIface.cc:120] void DBusMenuExporterPrivate::addAction(QAction*, int): Already tracking action "&View" under id 64
[Wrn] [GuiIface.cc:120] void DBusMenuExporterPrivate::addAction(QAction*, int): Already tracking action "&Window" under id 68
[Wrn] [GuiIface.cc:120] void DBusMenuExporterPrivate::addAction(QAction*, int): Already tracking action "&Help" under id 73
[Wrn] [Publisher.cc:141] Queue limit reached for topic /gazebo/default/user_camera/pose, deleting message. This warning is printed only once.
```

Gazebo tries find ros files, but there is no more. By the way, by installing the gazebo proposed by the example, it deletes before libraries "libgazebo_ros_paths_plugin.so" and "libgazebo_ros_api_plugin.so".

Even if I do not want to use Gazebo, for this example, I do not even know how to create a simple example of publisher and subscriber with orocos_rtt.

ahoarau, 2018-09-04:
You did not have to follow the whole instruction, just the Orocos + rtt-ros-integration part.
Gazebo is obviously not needed.

You just need to execute : 
```bash
sudo sh -c "echo 'deb http://packages.ros.org/ros/ubuntu $(lsb_release -cs) main' > /etc/apt/sources.list.d/ros-latest.list"
wget http://packages.ros.org/ros.key -O - | sudo apt-key add -
sudo apt update
sudo apt install python-rosdep python-catkin-tools ros-kinetic-catkin python-wstool python-vcstool
```

```bash
mkdir ~/isir ; cd ~/isir
mkdir -p orocos-2.9_ws/src
cd orocos-2.9_ws/src
# Get all the packages
wstool init
wstool merge https://raw.githubusercontent.com/kuka-isir/rtt_lwr/rtt_lwr-2.0/lwr_utils/config/orocos_toolchain-2.9.rosinstall
wstool update -j2
# Get the latest updates (OPTIONAL)
cd orocos_toolchain
git submodule foreach git checkout toolchain-2.9
git submodule foreach git pull
# Configure the workspace
cd ../../
# Install dependencies
source /opt/ros/kinetic/setup.bash
rosdep install --from-paths src --ignore-src --rosdistro kinetic -y -r
catkin config --init --install --extend /opt/ros/kinetic/ --cmake-args -DCMAKE_BUILD_TYPE=Release
# Build
catkin build
```

and 
```bash
cd ~/isir
mkdir -p rtt_ros-2.9_ws/src
cd rtt_ros-2.9_ws/src
# Get all the packages
wstool init
wstool merge https://github.com/kuka-isir/rtt_lwr/raw/rtt_lwr-2.0/lwr_utils/config/rtt_ros_integration-2.9.rosinstall
wstool update -j2
# Configure the workspace
cd ../
# Install dependencies
source ~/isir/orocos-2.9_ws/install/setup.bash
rosdep install --from-paths src --ignore-src --rosdistro kinetic -y -r
catkin config --init --install --extend ~/isir/orocos-2.9_ws/install --cmake-args -DCMAKE_BUILD_TYPE=Release
# Build (this can take a while)
catkin build
```

ahoarau, 2018-09-04:
Note that these instructions are for orocos 2.9.

Please remove any orocos debian before running those commands : `sudo apt remove ros-kinetic-rtt*`

MatheusPinto, 2018-09-04:
I see... I just can not figure out how to create a simple package with files to publish and subscribe, so I can test.

ahoarau, 2018-09-04:
Some examples can be found here : https://github.com/jhu-lcsr/rtt_ros_examples

MatheusPinto, 2018-09-07:
I need some more explanatory examples. I do not even know in which directory to put these folders. Also, I do not see the relationship between publishers and subscribers in these examples.

ahoarau, 2018-09-07:
Orocos RTT documentation is here : https://orocos-toolchain.github.io/rtt/toolchain-2.9/xml/orocos-components-manual.html

Once you have an port of a ros type you can stream (input/output) like this https://github.com/jhu-lcsr/rtt_ros_examples/blob/master/rtt_ros_integration_example/example.ops#L11

that you launch using `deployer -s example.ops`

MatheusPinto, 2018-09-08:
Thanks for the answers! However, the most basic thing I still dont know how to do: Any of the folders in this example (https://github.com/jhu-lcsr/rtt_ros_examples), should be placed in which directory? In the normal ROS catkin? Or what directory of the link installation you went through (https://rtt-lwr.readthedocs.io/en/latest/install/install-16.04-kinetic.html#orocos-2-9-rtt-ros-integration-2-9-from-source)?

closed 2018-09-09 by MatheusPinto