orocos/rtt_ros_integration #114: Problem to build orocos-rtt 2.9 for the xenomai 3 + ROS Kinetic

MatheusPinto, 2018-09-11:
Hi,

I follow the steps from here (https://github.com/orocos/rtt_ros_integration) and here (https://rtt-lwr.readthedocs.io/en/latest/rtpc/orocos-xenomai.html), to install orocos-rtt  in a Ubuntu Mate + Xenomai 3.0.6 environment in a Raspberry Pi 3.

```
# Compile for Xenomai
export OROCOS_TARGET=xenomai
# Create the workspace and clone orocos-toolchain
mkdir -p ~/ws/underlay_isolated/src/orocos
cd ~/ws/underlay_isolated
git clone --recursive https://github.com/orocos-toolchain/orocos_toolchain.git src/orocos/orocos_toolchain
catkin_make_isolated --install
```

However, in last command, to build, a error occurs when process the 'rtt' package:

```
make
Detected OROCOS_TARGET environment variable. Using: xenomai
-- CMAKE_VERSION: 3.5.1
Boost found in /usr/include
Orocos target is xenomai
XENOMAI_INCLUDE_DIR=XENOMAI_INCLUDE_DIR-NOTFOUND
XENOMAI_NATIVE_LIBRARY=XENOMAI_NATIVE_LIBRARY-NOTFOUND
CMake Error at config/LibFindMacros.cmake:74 (message):
  Required library XENOMAI NOT FOUND.

  Install the library (dev version) and try again.  If the library is already
  installed, set the XENOMAI_ROOT_DIR environment variable or use cmake to
  set the missing variables manually.
Call Stack (most recent call first):
  config/FindXenomai.cmake:72 (libfind_process)
  config/check_depend.cmake:147 (find_package)
  CMakeLists.txt:115 (INCLUDE)
```

It seems that he does not find xenomai libraries. However the "XENOMAI_NATIVE_LIBRARY" variable from the erros, the "native" caught my eye, because it refers to the "native" library from Xenomai 2 versions. It leads to believe that it seeks a Xenomai 2 version, not the Xenomai 3.

Looking the "FindXenomai.cmake" file from the error, he really searches for native library from Xenomai 2.

My question is: does orocos rtt have xenomai 3 support? If yes, how can I resolve this issue?

ahoarau, 2018-09-11:
Discussed here : https://github.com/orocos-toolchain/rtt/issues/263
I made a beta version of this support. not merged yet.

MatheusPinto, 2018-09-13:
@ahoarau I download your version and execute `catkin_make_isolated --install` from it. I had to do a swap of a pendrive, because at the time of installation the environment freezing near 10% (it overflow the limit of 1GB of raspberry memory). Putting a 2GB swap on the pendrive was enough. Installation required 1 GB of rpi memory + 1.5 GB of swap at most (analyzing on system monitor).
In the meantime, 3 installation attempts (each in an average of 4 hours) have been carried out and the installation is always stucked at 96%. Making a `ctrl+c`, this is returned from terminal:
```

==> Processing plain cmake package: 'rtt'
==> Building with env: '/home/raspberry/catkin_ws/underlay_isolated/install_isolated/env.sh'
Makefile exists, skipping explicit cmake invocation...
==> make cmake_check_build_system in '/home/raspberry/catkin_ws/underlay_isolated/build_isolated/rtt/install'
==> make -j7 in '/home/raspberry/catkin_ws/underlay_isolated/build_isolated/rtt/install'
Use: make docapi dochtml docpdf to build the documentation.
[  0%] Built target message
[ 60%] Built target orocos-rtt-xenomai_dynamic
[ 61%] Built target orocos-rtt-mqueue-xenomai_dynamic
[ 66%] Built target rtt-marshalling-xenomai_plugin
[ 67%] Building CXX object rtt/typekit/CMakeFiles/rtt-typekit-xenomai_plugin.dir/RealTimeTypekitTypes2.cpp.o
[ 67%] Building CXX object rtt/typekit/CMakeFiles/rtt-typekit-xenomai_plugin.dir/RealTimeTypekitTypes.cpp.o
[ 69%] Built target rtt-transport-mqueue-xenomai_plugin
[ 96%] Built target rtt-scripting-xenomai_plugin
^C^Xrtt/typekit/CMakeFiles/rtt-typekit-xenomai_plugin.dir/build.make:158: recipe for target 'rtt/typekit/CMakeFiles/rtt-typekit-xenomai_plugin.dir/RealTimeTypekitTypes.cpp.o' failed
make[2]: *** [rtt/typekit/CMakeFiles/rtt-typekit-xenomai_plugin.dir/RealTimeTypekitTypes.cpp.o] Interrupt
rtt/typekit/CMakeFiles/rtt-typekit-xenomai_plugin.dir/build.make:182: recipe for target 'rtt/typekit/CMakeFiles/rtt-typekit-xenomai_plugin.dir/RealTimeTypekitTypes2.cpp.o' failed
make[2]: *** [rtt/typekit/CMakeFiles/rtt-typekit-xenomai_plugin.dir/RealTimeTypekitTypes2.cpp.o] Interrupt
CMakeFiles/Makefile2:774: recipe for target 'rtt/typekit/CMakeFiles/rtt-typekit-xenomai_plugin.dir/all' failed
make[1]: *** [rtt/typekit/CMakeFiles/rtt-typekit-xenomai_plugin.dir/all] Interrupt
Makefile:138: recipe for target 'all' failed
make: *** [all] Interrupt
Traceback (most recent call last):
  File "/opt/ros/kinetic/bin/catkin_make_isolated", line 160, in <module>
    main()
  File "/opt/ros/kinetic/bin/catkin_make_isolated", line 156, in main
    override_build_tool_check=opts.override_build_tool_check,
  File "/opt/ros/kinetic/lib/python2.7/dist-packages/catkin/builder.py", line 981, in build_workspace_isolated
    number=index + 1, of=len(ordered_packages)
  File "/opt/ros/kinetic/lib/python2.7/dist-packages/catkin/builder.py", line 691, in build_package
    destdir=destdir, use_ninja=use_ninja
  File "/opt/ros/kinetic/lib/python2.7/dist-packages/catkin/builder.py", line 544, in build_cmake_package
    run_command(make_cmd, build_dir, quiet)
  File "/opt/ros/kinetic/lib/python2.7/dist-packages/catkin/builder.py", line 239, in run_command
    proc.wait()
  File "/usr/lib/python2.7/subprocess.py", line 1392, in wait
    pid, sts = _eintr_retry_call(os.waitpid, self.pid, 0)
  File "/usr/lib/python2.7/subprocess.py", line 476, in _eintr_retry_call
    return func(*args)
KeyboardInterrupt
```

Is it a deadlock?