orocos/rtt_ros_integration #118: Equivalent script "stream" function in C++

MatheusPinto, 2019-02-20:
Hello,

I am downloaded the examples provided in here to use Orocos+ROS: https://github.com/jhu-lcsr/rtt_ros_examples.

I achieved to run the example.ops file, however I want to run directly in C++ (not in a script).

So, I created the follow code:
```

#include <rtt/TaskContext.hpp>
#include <rtt/Port.hpp>
#include <std_msgs/Float64.h>
#include <std_msgs/String.h>
#include <rtt/Component.hpp>
#include <std_srvs/Empty.h>
#include <rtt/Activity.hpp> // for use setActivity
#include <rtt/os/main.h> // for linker find ORO_main function
#include <ros/ros.h>
//#include <iostream>

using namespace RTT;

class HelloRobot : public RTT::TaskContext
{

private: 
  InputPort<std_msgs::Float64> inport;
  OutputPort<std_msgs::Float64> outport;

  InputPort<std_msgs::String> sinport;
  OutputPort<std_msgs::String> soutport;
 
  std::string prop_answer;
  double prop_counter_step;

  double counter;
 
public:

  HelloRobot(const std::string& name):
    TaskContext(name),
    inport("float_in"),outport("float_out"),
    sinport("string_in"),soutport("string_out","Hello Robot"),
    prop_answer("Hello Robot"),prop_counter_step(0.01), counter(0.0)
  {
    this->addEventPort(inport).doc("Receiving a message here will wake up this component.");
    this->addPort(outport).doc("Sends out 'answer'.");
    this->addEventPort(sinport).doc("Receiving a message here will wake up this component.");
    this->addPort(soutport).doc("Sends out a counter value based on 'counter_step'.");
  }

  ~HelloRobot(){}

private:

  void updateHook()
  {
    std_msgs::Float64 fdata;
    std_msgs::String sdata;

    //std::cout << "hello" << std::endl;

    if(NewData==inport.read(fdata))
    {
      log(Info)<<"Float in: "<<fdata<<endlog();
    }

    if(NewData==sinport.read(sdata))
    {
      log(Info)<<"String in: "<<sdata<<endlog();
    }
    counter+=prop_counter_step;
    fdata.data+=counter;
    outport.write(fdata);
    sdata.data=prop_answer;
    soutport.write(sdata);
   }

};
ORO_CREATE_COMPONENT(HelloRobot)

int ORO_main(int argc, char** argv)
{
   // Setup a component
   HelloRobot myComponent("HelloRobot");
   // Execute a component
   myComponent.setActivity( new Activity(HighestPriority, 0.01 ) );
   myComponent.start();
  
   // Wait when task are not executing anymore
   getchar();

   // Finalize component
   myComponent.stop();

   return 0;
}
```

I can generated the executable. However, I dont find in any Orocos document the equivalent in C++ of the function "stream" for script, to connect a port to a ROS topic.
Do you have this information to give?

meyerj, 2019-02-27:
There are two aspects here:

1. You have to first load the typekit plugins for `std_msgs`, which is `rtt_std_msgs`. You can use [rtt_ros](https://github.com/orocos/rtt_ros_integration/tree/toolchain-2.9/rtt_ros) to import all plugins provided by a ROS package and all of its plugin dependencies. These also include `rtt_roscomm` and `rtt_rosnode`, which actually makes your process a ROS node.

   ```cpp
   #include <rtt_ros/rtt_ros.h>

   int ORO_main(int argc, char** argv)
   {
     rtt_ros::import("rtt_std_msgs");
     // ...implicitly imports rtt_roscomm and rtt_rosnode.

     // ...
   }
   ```

2. You can create a streaming connection to or from a ROS topic using [RTT::base::PortInterface::createStream()](https://orocos-toolchain.github.io/rtt/master/api/html/classRTT_1_1base_1_1PortInterface.html#a1599147a15dbfbb62defbee0c51b65f6) and one the helpers in package [rtt_roscomm](https://github.com/orocos/rtt_ros_integration/blob/jade-devel/rtt_roscomm/include/rtt_roscomm/rtt_rostopic.h):

   ```cpp
   #include <rtt_roscomm/rtt_rostopic.h>

   // at some point, e.g. in ORO_main() or in the constructor of HelloRobot:
   myComponent->getPort("inport")->createStream(rtt_rostopic::topic("float_in"));
   myComponent->getPort("inport")->createStream(rtt_rostopic::topic("float_out"));
   myComponent->getPort("sinport")->createStream(rtt_rostopic::topic("string_in"));
   myComponent->getPort("soutport")->createStream(rtt_rostopic::topic("string_out"));

   // Should check return values of createStream() because it could fail, e.g. if the transport plugin
   // has not been loaded before.
   ```

   `topicBuffer()` instead of `topic()` creates a buffer connection which lowers the risk of missing messages if you publish too fast or do not read from the port fast enough.

MatheusPinto, 2019-02-28:
Thanks @meyerj, that works!

closed 2019-02-28 by MatheusPinto