orocos/rtt_ros_integration #161: Typegen - the last frontier

neuebot, 2020-11-28:
Dear all,

I write you this email as a last effort, to try to solve a problem I have with typegen. 
I am trying to publish a paper to introduce a solution I developed using Orocos (https://github.com/neuebot/NeueBot), but I am struggling with the setup.

Long story short, I implemented and tested my setup in Ubuntu 14.04 with Orocos Toolchain-2.8.0 compiled from source. The setup was a struggle also back then, but I managed to have everything running, including the CMakeLists macro orogen_typegen_headers.

I think the typekits are rather simple... Here is one,
```cpp
#ifndef TEST_SUPERVISOR_TYPES_HPP
#define TEST_SUPERVISOR_TYPES_HPP

#include <kdl/frames.hpp>

#include <string>
#include <vector>

namespace RTT {
    struct vector_int {
        std::vector< int > data;
    };

    struct vector_frame {
        std::vector< KDL::Frame > data;
    };

    struct vector_vector_double {
        std::vector< std::vector< double > > data;
    };

    struct redundancy_scores {
        int res;
        std::vector< int > GC;
        std::vector< std::vector< double > > PsiIntervals;
    };

    struct error_msg {
        int level;
        std::string msg;
    };

    struct posture {
        KDL::Frame frame;
        double psi;
        unsigned int GC;
    };
}

#endif // TEST_SUPERVISOR_TYPES_HPP
```

I was happy to find the new documentation at https://docs.orocos.org/index.html, and I found out how easy it was to set up the system using ROS debian packages (Ubuntu 16.04). It felt like a cheat.

The problem is that I use my own typekits and that functionality is still not supported.
I am trying every way I can to fit this last piece of the puzzle, I am testing on like 5 different virtual machines with clean installations at a time. 

I tried installing different orocos toolchain versions from source, I tried following this guide https://rtt-lwr.readthedocs.io/en/latest/install/install-16.04-kinetic.html, I tried using ros packages. I have ran down more rabbit holes than I can count :sob:

Can you shed some light on how to generate my typekits for a simple struct like the one above?

caplett, 2020-12-01:
hey @neuebot i just started evaluating orocos so i can not give indepth help at all.
i've found this page: https://orocos.github.io/rtt/toolchain-2.9/xml/orocos-typekit-plugin.html
maybe it leads in a direction which could help.

good luck!

neuebot, 2020-12-01:
Hello @caplett!

Thanks for replying. 
I worked with Orocos for some time now, I think it is a fantastic solution, the only problem is that it is hard to maintain. 

I have come across that manual before. I tried following it to the line, but I had no success...
1) I tried compiling the toolchain 2.9 from source, and use the typegen as they suggest in sec. 1, but I get the following error,

```sh
-- Orocos-RTT found in /home/neuebot/orocos_toolchain/install/lib/cmake/orocos-rtt/orocos-rtt-gnulinux-libraries.cmake
-- - Detected OROCOS_TARGET environment variable. Using: gnulinux
-- Found orocos-rtt 2.9.0 for the gnulinux target. 
-- - Found requested orocos-rtt components: rtt-marshalling
-- [UseOrocos] Building package cartesian_trajectory_controller without an external buildtool like rosbuild or catkin
-- [UseOrocos] Using Orocos RTT in cartesian_trajectory_controller
-- [orocos_get_manifest_deps] Note: this package has no manifest.xml file. No dependencies can be auto-configured.
-- [UseOrocos] Building library CartesianTrajectory
-- [UseOrocos] Building component cartesian_trajectory_controller in library cartesian_trajectory_controller-gnulinux
-- [UseOrocos] Generating typekit for cartesian_trajectory_controller ...
Cannot require 'typelib'
If you are using Rock, the 'typelib' package should have been installed automatically.
First, make sure that you have loaded autoproj's env.sh script before continuing
Second, typelib should be installed in tools/typelib from the root of your Rock installation.
If it is not the case, report this to the rock developers. To workaround, run
  amake typelib
and try again.
CMake Error at /home/neuebot/orocos_toolchain/install/lib/cmake/orocos-rtt/UseOROCOS-RTT.cmake:611 (add_subdirectory):
  The source directory

    /home/neuebot/neuebot/neuebot_ws/build/cartesian_trajectory_controller/typekit

  does not contain a CMakeLists.txt file.
Call Stack (most recent call first):
  CMakeLists.txt:27 (orocos_typegen_headers)

-- [UseOrocos] Generating package version 1.0 (default version).
-- [UseOrocos] Exporting libraries cartesian_trajectory_controller-typekit.
-- [UseOrocos] Exporting include directories /home/neuebot/neuebot/neuebot_ws/build/cartesian_trajectory_controller/typekit.
-- [UseOrocos] Exporting library directories /usr/local/lib/orocos/gnulinux/cartesian_trajectory_controller/types.
-- Configuring incomplete, errors occurred!
See also "/home/neuebot/neuebot/neuebot_ws/build/cartesian_trajectory_controller/CMakeFiles/CMakeOutput.log".
```
But this issue is already reported at https://github.com/orocos-toolchain/orocos_toolchain/issues/39.

I opened a new one because I failed to find a way to get this to work no matter how many different solutions I tried (I followed this rabbit hole too -> https://gitlab.syscop.de/TOPCORE/topcore_orocos_packages/-/issues/8).

About the other sections (2 and 3), I cannot figure them out... I tried to generate a typekit for a simple example with no success. Either things are out-of-date, or I don't know what exactly what to do. I tried to look for examples of typekits generated this way, with no success also.

cf-dtx, 2020-12-19:
Hi @psoetens and @smits 
I am sorry to bother you, but can you shed some light on this topic?
I couldn't make any progress 🤔

meyerj, 2021-01-27:
Sorry for the late reply. But as this issue is unrelated to rtt_ros_integration, let's continue in https://github.com/orocos-toolchain/orocos_toolchain/issues/39.

closed 2021-01-27 by meyerj